strongtz/i915-sriov-dkms #334: unknown parameter 'max_vfs' ignored

ParticleG, 2025-08-15:
I've installed [i915-sriov-dkms](https://github.com/strongtz/i915-sriov-dkms) from AUR, and enabled the kernel parameters in `/etc/kernel/cmdline` (Since I'm using a UKI). I've installed sysfsutils and configs one vfs in `/etc/sysfs.conf`. But after reboting the system, I could only find one VGA device in PCI list, also content of `/sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs` is 0.

```bash
➜  ~ git:(main) ✗ uname -a
Linux particleg-t14pg3 6.16.0-arch2-1 #1 SMP PREEMPT_DYNAMIC Wed, 13 Aug 2025 23:38:48 +0000 x86_64 GNU/Linux
➜  ~ git:(main) ✗ sudo cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 1
➜  ~ git:(main) ✗ sudo dmesg | grep i915  
[    0.000000] Command line: root=PARTUUID=ce4b9833-e845-4824-b0d5-5e9daff20595 zswap.enabled=0 rootflags=subvol=@ rw rootfstype=btrfs resume=UUID=34594840-1534-4d26-9558-2cc4d66b70a7 resume_offset=1352960 intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    0.069056] Kernel command line: root=PARTUUID=ce4b9833-e845-4824-b0d5-5e9daff20595 zswap.enabled=0 rootflags=subvol=@ rw rootfstype=btrfs resume=UUID=34594840-1534-4d26-9558-2cc4d66b70a7 resume_offset=1352960 intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    1.414879] i915: unknown parameter 'max_vfs' ignored
[    1.415100] i915 0000:00:02.0: [drm] Found meteorlake (device ID 7d51) integrated display version 14.00 stepping D0
[    1.417357] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.442819] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.442848] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.455238] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.463908] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/mtl_dmc.bin (v2.23)
[    1.501828] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin version 70.44.1
[    1.518987] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    1.518992] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    1.519220] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    1.528242] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.44.1
[    1.528247] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
[    1.551135] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for clear media
[    1.552427] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
[    1.552429] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
[    1.552498] i915 0000:00:02.0: [drm] GT1: GUC: RC enabled
[    1.556171] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    1.558513] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 1
[    1.613097] fbcon: i915drmfb (fb0) is primary device
[    1.662673] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    1.699710] i915 0000:00:02.0: [drm] GT1: Loaded GSC firmware i915/mtl_gsc_1.bin (cv1.0, r102.1.15.1926, svn 1)
[    1.720079] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for all workloads
[    3.922133] mei_gsc_proxy 0000:00:16.0-0f73db04-97ab-4125-b893-e904ad0d5464: bound 0000:00:02.0 (ops i915_gsc_proxy_component_ops [i915])
[    4.243271] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops intel_audio_component_bind_ops [i915])
[  355.864698] i915 0000:00:02.0: Using 41-bit DMA addresses
```

I've noticed the dmesg has a line saying `i915: unknown parameter 'max_vfs' ignored`. Is this the cause? Is the 6.16.0 kernel version not supported?

johntdavis84, 2025-08-15:
@ParticleG How did you resolve this?

ParticleG, 2025-08-16:
> [@ParticleG](https://github.com/ParticleG) How did you resolve this?

I just figured out I installed the package without having the `linux-headers` package on my machine. After installing linux-headers and reinstalling the i915-sriov-dkms package, the module was successfully installed. But I still can't set numvfs for my iGPU

ParticleG, 2025-08-16:
After installing dkms module, I reboot the system and print some logs:
``` bash
➜  ~ git:(main) ✗ sudo cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 2
➜  ~ git:(main) ✗ sudo cat /etc/kernel/cmdline 
root=PARTUUID=ce4b9833-e845-4824-b0d5-5e9daff20595 zswap.enabled=0 rootflags=subvol=@ rw rootfstype=btrfs resume=UUID=34594840-1534-4d26-9558-2cc4d66b70a7 resume_offset=1352960 intel_iommu=on iommu=pt
➜  ~ git:(main) ✗ sudo dmesg | grep i915      
[    0.000000] Command line: root=PARTUUID=ce4b9833-e845-4824-b0d5-5e9daff20595 zswap.enabled=0 rootflags=subvol=@ rw rootfstype=btrfs resume=UUID=34594840-1534-4d26-9558-2cc4d66b70a7 resume_offset=1352960 intel_iommu=on iommu=pt i915.enable_guc=3 i915.force_probe=7d51 i915.max_vfs=2 module_blacklist=xe
[    0.068848] Kernel command line: root=PARTUUID=ce4b9833-e845-4824-b0d5-5e9daff20595 zswap.enabled=0 rootflags=subvol=@ rw rootfstype=btrfs resume=UUID=34594840-1534-4d26-9558-2cc4d66b70a7 resume_offset=1352960 intel_iommu=on iommu=pt i915.enable_guc=3 i915.force_probe=7d51 i915.max_vfs=2 module_blacklist=xe
[    3.831169] i915: loading out-of-tree module taints kernel.
[    3.831226] i915: module verification failed: signature and/or required key missing - tainting kernel
[    4.014447] i915: You are using the i915-sriov-dkms module, a ported version of the i915 module with SR-IOV support.
[    4.014448] i915: Please file any bug report at https://github.com/strongtz/i915-sriov-dkms/issues/new.
[    4.014448] i915: Module Homepage: https://github.com/strongtz/i915-sriov-dkms
[    4.014544] i915 0000:00:02.0: Force probing unsupported Device ID 7d51, tainting kernel
[    4.014589] i915 0000:00:02.0: [drm] Found METEORLAKE (device ID 7d51) display version 14.00 stepping D0
[    4.014596] i915 0000:00:02.0: Running in SR-IOV PF mode
[    4.016690] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.067255] i915 0000:00:02.0: vgaarb: deactivate vga console
[    4.067286] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    4.081214] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    4.095475] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/mtl_dmc.bin (v2.23)
[    4.149035] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin version 70.44.1
[    4.175374] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    4.175377] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    4.175623] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    4.186290] mei_gsc_proxy 0000:00:16.0-0f73db04-97ab-4125-b893-e904ad0d5464: bound 0000:00:02.0 (ops i915_gsc_proxy_component_ops [i915])
[    4.186750] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.44.1
[    4.186754] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
[    4.274136] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for clear media
[    4.277334] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
[    4.277347] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
[    4.277593] i915 0000:00:02.0: [drm] GT1: GUC: RC enabled
[    4.290986] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    4.305215] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 1
[    4.434075] i915 0000:00:02.0: [drm] GT1: Loaded GSC firmware i915/mtl_gsc_1.bin (cv1.0, r102.1.15.1926, svn 1)
[    4.454602] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for all workloads
[    4.459827] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.459979] fbcon: i915drmfb (fb0) is primary device
[    4.465316] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    4.480752] i915 0000:00:02.0: 2 VFs could be associated with this PF
➜  ~ git:(main) ✗ sudo echo 2 > /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs 
zsh: 权限不够: /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs
➜  ~ git:(main) ✗ sudo echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs          
zsh: 权限不够: /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
➜  ~ git:(main) ✗ echo 2 > sudo tee /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
➜  ~ git:(main) ✗ cat /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs 
0
➜  ~ git:(main) ✗ cat /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_totalvfs 
2
➜  ~ git:(main) ✗ echo 2 > sudo tee /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs 
➜  ~ git:(main) ✗ cat /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_totalvfs            
2
➜  ~ git:(main) ✗ cat /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs              
0
➜  ~ git:(main) ✗ sudo cat /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs
0
```
`sriov_totalvfs` is 2, but I couldn't set sriov_numvfs

ParticleG, 2025-08-16:
Okay, never mind. `sudo su` switched to root user, just solved it. Don't know why `sudo echo` not working

closed 2025-08-16 by ParticleG