strongtz/i915-sriov-dkms #335: Windows Guest - Error 43

lutfijw, 2025-08-17:
Hi All,
I am using Intel Arrow Lake CPU and tried installing the i915-SRIOV-DKMS module in proxmox and managed to see 7 VF.
I have no issue in assigning the VF to Linux VM, however when I assign the VF to WIndows VM, I always see error 43 in device manager and no GPU identified.

Please help to provide guidance on how to properly assign the VF to Windows guest VM in proxmox.

Regards,
Lutfi

closed 2025-08-19 by lutfijw

lutfijw, 2025-08-19:
> Glad you were able to identify the problem

Thank you for great support. See below screenshot confirming GPU is working

[image]

johntdavis84, 2025-08-20:
Congratulations. It's such a game changer once this is working. :)

YehezkielLegowo, 2025-08-25:
Hello, can you please share the tips or how to use it, because i'am on PVE 9 and encounter VF problems :(

lutfijw, 2025-08-26:
> Hello, can you please share the tips or how to use it, because i'am on PVE 9 and encounter VF problems :(

How far have you gone and what error you have?

uzumo, 2025-08-26:
I'm a little tired so I'd be happy if you could help him out.

lutfijw, 2025-08-27:
> I'm a little tired so I'd be happy if you could help him out.

Buddy, after further test seem like the Windows VM is struggling to play any H264 video. With small resolution, I still got stuttering issue and when I check at host (proxmox), it is showing GPU HANG error, but since the resolution is small, proxmox manage to reset the VF.

Not sure whether the error is caused by the DKMS driver or other things.

Something just missing. FYI, I am also running Frigate in VM and found the inference time sometime goes up to 40ms. Compared to native GPU, the vGPU inference time is 4 times higher.

Any idea what caused the issue?

[image]

uzumo, 2025-08-27:
Perhaps it is due to a difference in my setup or computer configuration.
Let's move on to another thread.