strongtz/i915-sriov-dkms #337: Blank Video - Arrow Lake iGPU (Black Screen)

lutfijw, 2025-08-21:
Hi All,
I managed to install the i915 DKMS on proxmox VE 9 and able to passthrough the iGPU VF to Windows VM based on guidance from @uzumo at [https://github.com/strongtz/i915-sriov-dkms/issues/335](url) with no error and GPU is recognized on Windows.

While Windows recognized the GPU, I can't play Video. Video is played, audio is coming out, however the screen just shown black.

On task manager, I can see high jump in GPU utilization when Video is played, however only for few seconds then the spike, return back to 1% load. I can confirm the video file is good as I can play it outside the VM.

I did try to play the Video using VNC, Windows Media Player in the VM, however the result is the same, audio played, no video.

Appreciate your guidance on what caused this issue.

[image]

Regards,

johntdavis84, 2025-08-21:
What kind of video is it?

If you're seeing activity on the GPU, and hearing audio on the TS file _and_ have working YouTube playback, I think the i915 DKMS driver on the host and Xe driver inside the VM are working well.

This sounds like a codec issue inside the VM. That is, like Windows needs some additional software installed to know how to play the video. That's ... 4K HDR, in a .ts file? That's a TransportStream file, so it's a container for another video codec (I think).

What does VLC say when you pull up information about the video file while it's open and playing? It should be able to tell you the video and audio codecs being used.

Also, try Windows Media Player and see if that works.

lutfijw, 2025-08-21:
Seem like the issue impacting any video file. I tried everything, video downloaded from internet, video taken from my mobile, low resolution video and all end up the same, audio coming out but no video.
As per my earlier screenshot, the GPU seem only activated for few seconds, then no activity.

lutfijw, 2025-08-21:
I tried restarting the proxmox host and now seem like everything are working.
Not really sure what was happened before.

[image]

lutfijw, 2025-08-22:
I am able to simulate the issue. Seem like the driver is crash at host when I am playing 8k Video in Windows VM. The Windows guest is not aware that the host i915 GPU driver is crashed and continue to behave normally but the video is showing black screen.
You can try to replicate in your box by downloading 8k Video and play the video in Windows VM, the VM will initially struggle to play the video and after few minutes it display will turn into black. Disconnect the Remote Desktop and reconnect again, everything will seem normal, no error shown, but you wont be able to play any video, regardless the resolution or file type.

Restarting VM is not fixing the issue. In the host, I see some many timeout notification when I run dmesg |grep i915, see below:

[image]

lutfijw, 2025-08-22:
Is there anyway to restart the i915 driver without rebooting the host? I think if we can make it auto restart, then it will fix the issue but the driver will still struggle to play 8k video.

uzumo, 2025-08-22:
It certainly seems that this error is the cause.

However, I'm not an expert, so I don't know how to solve it.

I tried to check the reproducibility at 8k, but I could only get up to 4k.

Sorry.

lutfijw, 2025-08-22:
I can confirm the issue is related to i915 DKMS driver loaded in Proxmox host. When I play 8k Video on Windows 11 guest, I can see "GPU HANG" message on proxmox host, post that, no Video playable in Windows 11 guest. VM restart also not fixing the issue and I need to restart the host.

My system detail is as follow:

Processor: Intel Ultra Core 5 Arrow Lake 235
GPU: Intel iGPU (Intel Graphic)
Host OS: Proxmox VE 9 - Kernel 6.14.8-2-pve
i915 Version: i915-sriov-dkms_2025.07.22_amd64.deb 

/etc/kernel/cmdline
`root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on iommu=pt module_blacklist=xe i915.enable_guc=3 i915.max_vfs=7 i915.force_probe=7d67`

dmesg | grep i915
Showing so many fence expiration time out message.
`[13574.149826] Fence expiration time out i915-0000:00:02.0:av:h264:df0[292367]:2e!`
`[13592.534260] Fence expiration time out i915-0000:00:02.0:dec0:0:h264[293637]:2!`
`[13592.603155] Fence expiration time out i915-0000:00:02.0:dec0:0:h264[293637]:4!`

Hope someone can patch the bug.

uzumo, 2025-08-22:
I don't know if it's relevant, but h264 doesn't mention anything about 8k.

The log said h264, so I assumed it was h264.

https://edc.intel.com/content/www/jp/ja/design/ipla/software-development-platforms/client/platforms/alder-lake-desktop/12th-generation-intel-core-processors-datasheet-volume-1-of-2/004/hardware-accelerated-video-decode/

lutfijw, 2025-08-22:
> I don't know if it's relevant, but h264 doesn't mention anything about 8k.
>
> The log said h264, so I assumed it was h264. [...]

Yes, it was h264 at 8k. The GPU is hang and caused the issue. The driver it self should restrict playing then and throw an error. Today, the driver just stalled and system reboot is required to reload the driver.

Post reboot, everything are working as BAU, I can play video in VM, but only up to 4k resolution.

lutfijw, 2025-08-22:
Closing this and move to [338](https://github.com/strongtz/i915-sriov-dkms/issues/338)

uzumo, 2025-08-22:
Are you going to run the following command during the issue and improve it the next time the virtual machine starts without rebooting the host?

‘‘‘
echo "0000:00:02.1" > /sys/bus/pci/drivers/vfio-pci/unbind 2>/dev/null
echo "0000:00:02.1" > /sys/bus/pci/drivers/i915/bind 2>/dev/null
‘‘‘

※ Please do not execute at 0000:00:02.0 under any circumstances.

If this command is successful, it can resolve the issue without the need to reboot the host, even if the 8k h264 video is playing, by executing these commands through the hook script.

uzumo, 2025-08-22:
When executed at 0000:00:02.0, all VFs will disappear. 
If you are going to execute it at 0000:00:02.0, you will need the commands to create the VFs.

```
echo "0000:00:02.0" > /sys/bus/pci/drivers/vfio-pci/unbind 2>/dev/null
echo "0000:00:02.0" > /sys/bus/pci/drivers/i915/bind 2>/dev/null
echo 7 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
```

If it can be fixed here, it cannot be executed with hookscript, so manual execution is necessary.

lutfijw, 2025-08-22:
> When executed at 0000:00:02.0, all VFs will disappear. If you are going to execute it at 0000:00:02.0, you will need the commands to create the VFs.
>
> ```
> echo "0000:00:02.0" > [...]

Try to download this and play in your VM. The file was downloaded from internet and only demo file.

https://cloud.widayanto.net/s/RaYB3EFFsqsrHkj

uzumo, 2025-08-22:
When playing with VLC on an N100 computer, the CPU performance is too low, resulting in a slideshow effect. 
CPU usage is 100%.

Although the image quality was poor on mcp-be, it was played back using the GPU.

However, there are no errors in dmesg.

It might be because the performance is too low...

It's Proxmox 8, but I'll try it with 265k.

uzumo, 2025-08-22:
It is set up like my computer.

Guest OS
```
Windows 11 Pro : August 12, 2025—KB5063878 (OS Build 26100.4946)
```

GPU Driver
```
32.0.101.6913
```

Player
```
VLC 3.0.21
```

Local Group Policy
```
Settings - Policies - Administrative Templates - Windows Components/Remote Desktop Services/Remote Desktop Session Host/Remote Session Environment

Configure H.264/AVC hardware encoding for Remote Desktop Connections Enabled
Prioritize H.264/AVC 444 graphics mode for Remote Desktop Connections	Enabled
Use hardware graphics adapters for all Remote Desktop Services sessions	Enabled
```

Proxmox VE
```
root@pve:~# proxmox-boot-tool kernel list
Manually selected kernels:
None.

Automatically selected kernels:
6.14.8-2-bpo12-pve
6.14.8-2-pve
6.8.12-13-pve

Pinned kernel:
6.14.8-2-pve
```

```
root@pve1:~# dpkg -l | grep i915
ii  i915-sriov-dkms                             2025.07.22                          amd64        Linux i915 driver (dkms module) with SR-IOV support
```

```
root@pve1:~# ls -al /usr/share/kvm/arrowlake_igpu_265k.bin
-rwxrwxrwx 1 root root 140800 Jul 23 00:24 /usr/share/kvm/arrowlake_igpu_265k.bin
```

```
root@pve1:~# cat /etc/kernel/cmdline
root=ZFS=rpool/ROOT/pve-1 boot=zfs quiet intel_iommu=on iommu=pt split_lock_detect=off
```

```
root@pve1:~# cat /etc/modprobe.d/*
# modprobe information used for DKMS modules
#
# This is a stub file, should be edited when needed,
# used by default by DKMS.
options i915 enable_guc=3
options i915 max_vfs=7
options i915 force_probe=7d67
options kvm ignore_msrs=1 report_ignored_msrs=0
options kvm-intel nested=Y
options nct6683 force=1
# This file contains a list of modules which are not supported by Proxmox VE

# nvidiafb see bugreport https://bugzilla.proxmox.com/show_bug.cgi?id=701
blacklist nvidiafb
options zfs zfs_arc_max=8589934592
```

```
root@pve1:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
```

```
root@pve1:~# cat /etc/pve/qemu-server/929.conf
agent: 1
args: -cpu host,hv_passthrough,level=35,+vmx
audio0: device=ich9-intel-hda,driver=none
balloon: 0
bios: ovmf
boot: order=ide0;ide2;scsi0
cores: 20
cpu: host
efidisk0: zoi-client:vm-929-disk-0,efitype=4m,size=1M
hostpci0: 0000:00:02.1,pcie=1,romfile=arrowlake_igpu_265k.bin,x-vga=1
ide0: none,media=cdrom
ide2: none,media=cdrom
machine: pc-q35-9.2+pve1,viommu=intel
memory: 24576
meta: creation-qemu=9.0.0,ctime=1719914219
name: w11-ent-eval-24h2-temp
net0: virtio=BC:24:11:38:3B:23,bridge=vmbr0,firewall=1
numa: 0
ostype: win11
parent: setting
scsi0: zoi-client:vm-929-disk-1,backup=0,iothread=1,replicate=0,size=64G,ssd=1
scsi1: /dev/disk/by-id/scsi-200014fb188e4d25c,backup=0,iothread=1,replicate=0,size=112925M,ssd=1
scsihw: virtio-scsi-single
smbios1: uuid=85be5070-ece6-4dea-96d8-ff3877aeb867
sockets: 1
tags: default
vga: none
vmgenid: 0e6488d6-5ccf-4a3a-b701-79a9d151c8ec
```

lutfijw, 2025-08-22:
I also tried executing below command

```
echo "0000:00:02.7" > /sys/bus/pci/drivers/vfio-pci/unbind 2>/dev/null
echo "0000:00:02.7" > /sys/bus/pci/drivers/i915/bind 2>/dev/null
```

but I got below errors
```
[ 1752.375292] i915 0000:00:02.7: Force probing unsupported Device ID 7d67, tainting kernel
[ 1752.375331] i915 0000:00:02.7: [drm] Cannot read valid display IP version; disabling display.
[ 1752.375332] i915 0000:00:02.7: [drm] Found METEORLAKE (device ID 7d67) display version 0.00 stepping N/A
[ 1752.375337] i915 0000:00:02.7: Running in SR-IOV VF mode
[ 1752.375337] i915 0000:00:02.7: [drm] preliminary graphics version 12.70 media version 13.00
[ 1752.376961] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.20.1
[ 1752.377283] i915 0000:00:02.7: [drm] GT0: GMD_ID 0x3118004 version 12.70 step B0
[ 1752.379594] i915 0000:00:02.7: [drm] GT1: GUC: interface version 0.1.20.1
[ 1752.379670] i915 0000:00:02.7: [drm] GT1: GMD_ID 0x3400008 version 13.0 step C0
[ 1752.381769] i915 0000:00:02.7: [drm] VT-d active for gfx access
[ 1752.381786] i915 0000:00:02.7: [drm] Using Transparent Hugepages
[ 1752.382802] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.20.1
[ 1752.387773] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.20.1
[ 1752.388345] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[ 1752.388346] i915 0000:00:02.7: HuC firmware N/A
[ 1752.393487] i915 0000:00:02.7: [drm] GT1: GUC: interface version 0.1.20.1
[ 1752.395114] i915 0000:00:02.7: [drm] GT1: GUC: interface version 0.1.20.1
[ 1752.396711] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[ 1752.396712] i915 0000:00:02.7: HuC firmware PRELOADED
[ 1752.607108] i915 0000:00:02.7: [drm] CI tainted: 0x9 by intel_gt_set_wedged_on_init+0x34/0x50 [i915]
[ 1752.628375] i915 0000:00:02.7: [drm] *ERROR* Failed to initialize GPU, declaring it wedged!
[ 1752.628880] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628882] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628894] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628895] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628931] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628931] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628941] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628942] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628952] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628953] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628961] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628962] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628971] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628971] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628981] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628982] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.628990] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.628991] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629000] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629000] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629010] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629011] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629019] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629020] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629029] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629029] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629038] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629039] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629072] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629073] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629083] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629083] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629092] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629093] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629103] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629103] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[ 1752.629112] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0xf025005: failure a/0
[ 1752.629113] i915 0000:00:02.7: [drm] GT0: IOV: Failed to update VFs PTE by PF (-ENXIO)
[... cut ...]
[ 1800.817702] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11326]:4286!
[ 1820.758090] Fence expiration time out i915-0000:00:02.0:dec0:0:h264[56729]:f0!
[ 1840.719474] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11212]:366e!
[ 1860.668857] Fence expiration time out i915-0000:00:02.0:dec0:0:h264<57926>:3e!
[ 1880.634245] Fence expiration time out i915-0000:00:02.0:enc0:0:h264_vaa<58589>:2!
[ 1900.630631] Fence expiration time out i915-0000:00:02.0:dec0:0:h264<58906>:176!
```

lutfijw, 2025-08-22:
Below is the error happened when the GPU in host is hang:

```
[   60.579338] i915 0000:00:02.0: VF7 FLR
[  518.598677] i915 0000:00:02.0: VF7 FLR
[  519.666185] i915 0000:00:02.0: VF7 FLR
[  519.937524] i915 0000:00:02.0: VF7 FLR
[  754.815552] i915 0000:00:02.0: VF7 FLR
[  755.895120] i915 0000:00:02.0: VF7 FLR
[  756.150949] i915 0000:00:02.0: VF7 FLR
[  821.291800] i915 0000:00:02.0: VF7 FLR
[  822.359483] i915 0000:00:02.0: VF7 FLR
[  822.477632] i915 0000:00:02.0: VF7 FLR
[ 1355.703325] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:0:00000000
[ 1355.703385] i915 0000:00:02.0: [drm] GT1: Resetting chip for stopped heartbeat on vcs1
[ 1355.703666] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.44.1
[ 1355.703667] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
[ 1355.856482] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
[ 1355.856486] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
[ 1361.670557] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11212]:277e!
[ 1361.683185] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11326]:315e!
[ 1361.917076] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11212]:2780!
[ 1381.869174] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11326]:319c!
[ 1381.870460] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11326]:31a0!
[ 1381.870691] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11326]:319e!
[ 1381.873560] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11326]:31a2!
[ 1381.874467] Fence expiration time out i915-0000:00:02.0:av:h264:df0[11326]:31a4!
```

lutfijw, 2025-08-22:
> <img alt="Image" width="1179" height="2000" src="https://private-user-images.githubusercontent.com/65648539/480958513-a8fce1f9-c8dc-4d26-98c1-ecc5ce8e0472.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU4ODAzMzcsIm5iZiI6MTc1NTg4MDAzNywicGF0aCI6Ii82NTY0ODUzOS80ODA5NTg1MTMtYThmY2UxZjktYzhkYy00ZDI2LTk4YzEtZWNjNWNlOGUwNDcyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIyVDE2MjcxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUyMTljZjQ4MzMzNGE1NjYxNzAyMTFhNWUxNmQ5Yjc2NDJkODE3NmU5MDRkYzkxYzE2NDg0ZDkyMDNkMTk3OTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nlwM74pigWseF19b0ySLdopLdE0pIXQubylftlrY2kQ">
> I tried it at 265k, and I was able to confirm that the video was very clear and worked properly.
> 
> I will describe the settings here. Please provide the same settings once more.

Can you try to play till the video ended? If I open outside VM, there were several video sections in the file.
I was able to play for few minutes, but the section is not changing, the screen is frozen (same like your screenshot) but audio is keep playing.

uzumo, 2025-08-22:
>I got an error when starting the VM. The error caused due to enabling machine: pc-q35-9.2+pve1,viommu=intel
Removing the viommu=intel make VM start

Proxmox VE 8. This option is currently unavailable in Proxmox VE 9.

>I was able to play for few minutes, but the section is not changing, the screen is frozen (same like your screenshot) but audio is keep playing.

The screen freezes at 1:53.

It could be played over and over again.

[image]

However, there are still no errors in dmesg.

I don't know if there's a difference between Proxmox VE 8 and 9 because the 265k computer I use for work is not available for upgrade.

---
..I don't know why there are so many VLC processes.
I was scared when the sound continued even after I stopped the program, but it turns out the sound was being produced by the remaining processes.

In another program, the file plays without stopping, but there is noise.

[image]
[image]

lutfijw, 2025-08-27:
> I'm sorry I couldn't help you

Can help to run dxdiag and show the display and render?
Mine as follow:

[image]
[image]

lutfijw, 2025-08-27:
Thank you @uzumo, I can only check the BIOS on weekends, otherwise it will shutdown my entire home network. Many of my family members will complain.
Will try to look my BIOS setup and share the finding. Hoping it can help others

uzumo, 2025-08-27:
> Thank you @uzumo, I can only check the BIOS on weekends, otherwise it will shutdown my entire home network. Many of my family members will complain.

If you are using it as a server, it is normal that you cannot stop it, so there is no problem.
If possible, I would appreciate it if you could check by changing the playback software.

lutfijw, 2025-08-27:
> If you are using it as a server, it is normal that you cannot stop it, so there is no problem. If possible, I would appreciate it if you could check by changing the playback software.

Stupid of me, I found the issue. I did not blacklist the xe driver.
Somehow the i915 DKMS driver is still loaded, but not stable due to I did not blacklist the xe driver.

Once I blacklist the xe driver, everything run great…

Thank you so much for your great support!!!

uzumo, 2025-08-27:
I hope it worked. I'm happy too.

It seems that there is no need to blacklist Xe in the Proxmox VE 8 kernel, and I forgot to mention that. Sorry.
I learned that even if something isn't necessary, incorporating it can have a positive effect later on.

lutfijw, 2025-08-28:
> On the subject of performance loss, it appears that disabling viommu=intel will reduce performance. we have seen a performance loss of 8-18%.
> 
> There seems to be a workaround and after setting it, the performance is the same as pve8.
> 
> https://forum.proxmox.com/threads/vm-wont-start-with-pci-passthrough-after-upgrade-to-9-0.169586/post-795502

Enabling viommu=intel and adding below resolved the issue. I can now run two VM and both of it doing GPU processing.

qm set vmid -args '-cpu host,hv_passthrough,level=35,+vmx,guest-phys-bits=46 -global intel-iommu.aw-bits=39'