strongtz/i915-sriov-dkms #342: VFs get disabled after passing one through and starting VM

darkcrono, 2025-08-29:
Hello, I seem to be having an issue and I'm able to successfully see each pcie VFs, (00:02.0 - 00:02.3) but when I pass it through and start the VM (tried all excluding 02.0) I lose video from proxmox and i915 mesg shows the VFs get disabled.

Proxmox error shows,
`TASK ERROR: no pci device info for device '0000:00:02.1'`

while `dmesg | grep i915` shows,
```
[    7.910276] i915 0000:00:02.3: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    7.910279] i915 0000:00:02.3: HuC firmware PRELOADED
[    7.912012] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.912017] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[    7.912146] [drm] Initialized i915 1.6.0 for 0000:00:02.3 on minor 3
[    7.912311] i915 0000:00:02.0: Enabled 3 VFs
[   45.152186] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   45.152196] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   45.152200] i915 0000:00:02.3: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[   45.201151] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   45.201157] i915 0000:00:02.3: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   45.266135] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=io+mem:owns=io+mem
[   46.504037] i915 0000:00:02.0: Disabled 3 VFs
```

I'm on Proxmox 9.0.3 
uname -a
Linux pve 6.14.8-2-pve #1 SMP PREEMPT_DYNAMIC PMX 6.14.8-2 (2025-07-22T10:04Z) x86_64
dkms status
i915-sriov-dkms/2025.07.22, 6.14.8-2-pve, x86_64: installed (Original modules exist)

johntdavis84, 2025-08-29:
> darkcrono
>  created an issue

What’s the output of lspci -nnk ?

I’ve run two VMs concurrently before with no issues.

darkcrono, 2025-08-29:
```
#  lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:4617]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: igen6_edac
	Kernel modules: igen6_edac
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	DeviceName: Onboard - Video
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:02.1 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:02.2 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:02.3 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:02.4 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:02.5 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:02.6 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:02.7 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
	Kernel modules: xe, i915
00:0d.0 USB controller [0c03]: Intel Corporation Alder Lake-N Thunderbolt 4 USB Controller [8086:464e]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
00:14.0 USB controller [0c03]: Intel Corporation Alder Lake-N PCH USB 3.2 xHCI Host Controller [8086:54ed]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
00:14.2 RAM memory [0500]: Intel Corporation Alder Lake-N PCH Shared SRAM [8086:54ef]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:16.0 Communication controller [0780]: Intel Corporation Alder Lake-N PCH HECI Controller [8086:54e0]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: mei_me
	Kernel modules: mei_me
00:1c.0 PCI bridge [0604]: Intel Corporation Alder Lake-N PCI Express Root Port #1 [8086:54b8]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1c.2 PCI bridge [0604]: Intel Corporation Alder Lake-N PCI Express Root Port #3 [8086:54ba]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1c.3 PCI bridge [0604]: Intel Corporation Alder Lake-N PCI Express Root Port #4 [8086:54bb]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1d.0 PCI bridge [0604]: Intel Corporation Alder Lake-N PCI Express Root Port #9 [8086:54b0]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1d.1 PCI bridge [0604]: Intel Corporation Alder Lake-N PCI Express Root Port #10 [8086:54b1]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1f.0 ISA bridge [0601]: Intel Corporation Alder Lake-N PCH eSPI Controller [8086:5481]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Alder Lake-N PCH High Definition Audio Controller [8086:54c8]
	DeviceName: Onboard - Sound
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel, snd_soc_avs, snd_sof_pci_intel_tgl
00:1f.4 SMBus [0c05]: Intel Corporation Alder Lake-N SMBus [8086:54a3]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i801_smbus
	Kernel modules: i2c_i801
00:1f.5 Serial bus controller [0c80]: Intel Corporation Alder Lake-N SPI (flash) Controller [8086:54a4]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: intel-spi
	Kernel modules: spi_intel_pci
01:00.0 Ethernet controller [0200]: Aquantia Corp. AQC113C NBase-T/IEEE 802.3an Ethernet Controller [Marvell Scalable mGig] [1d6a:14c0] (rev 03)
	Subsystem: Aquantia Corp. Device [1d6a:0001]
	Kernel driver in use: atlantic
	Kernel modules: atlantic
02:00.0 SATA controller [0106]: ASMedia Technology Inc. ASM1164 Serial ATA AHCI Controller [1b21:1164] (rev 02)
	Subsystem: ASMedia Technology Inc. Device [1b21:2116]
	Kernel driver in use: ahci
	Kernel modules: ahci
03:00.0 SATA controller [0106]: ASMedia Technology Inc. ASM1164 Serial ATA AHCI Controller [1b21:1164] (rev 02)
	Subsystem: ASMedia Technology Inc. Device [1b21:2116]
	Kernel driver in use: ahci
	Kernel modules: ahci
04:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I226-V [8086:125c] (rev 04)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: igc
	Kernel modules: igc
05:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I226-V [8086:125c] (rev 04)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: igc
	Kernel modules: igc
```

current config is using 7 vfs now

darkcrono, 2025-08-29:
Also to note my `/etc/default/grub` file has,
`GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe"`

so xe should be blacklisted, not sure if it shows in the `lspci -nnk` makes a difference

altrnate32, 2025-09-01:
I have the same issue.
Strange thing is that I have an other server with almost the exact same hardware and same configuration (same kernel version also) and this server still works.
demsg from malfunctioning (note the last line):

```
[   14.248522] i915: You are using the i915-sriov-dkms module, a ported version of the i915 module with SR-IOV support.
[   14.248528] i915: Please file any bug report at https://github.com/strongtz/i915-sriov-dkms/issues/new.
[   14.248531] i915: Module Homepage: https://github.com/strongtz/i915-sriov-dkms
[   14.248838] i915 0000:00:02.0: [drm] Found ALDERLAKE_S (device ID 4680) display version 12.00 stepping C0
[   14.248862] i915 0000:00:02.0: Running in SR-IOV PF mode
[   14.249677] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   14.249682] i915 0000:00:02.0: vgaarb: deactivate vga console
[   14.249726] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[   14.250428] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[   14.259574] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[   14.265111] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.44.1
[   14.265119] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[   14.268218] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[   14.268599] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[   14.268602] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[   14.268988] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[   14.270365] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   14.270509] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[   14.270515] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   14.271417] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[   14.332831] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   14.332966] i915 0000:00:02.0: 7 VFs could be associated with this PF
[   14.362784] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   15.581067] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
[   15.581121] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   15.581143] i915 0000:00:02.1: [drm] Found ALDERLAKE_S (device ID 4680) display version 12.00 stepping C0
[   15.581165] i915 0000:00:02.1: Running in SR-IOV VF mode
[   15.582136] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.20.1
[   15.582824] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   15.582855] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   15.583249] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.20.1
[   15.583599] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.20.1
[   15.584115] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   15.584119] i915 0000:00:02.1: HuC firmware PRELOADED
[   15.586646] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   15.586650] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   15.586870] [drm] Initialized i915 1.6.0 for 0000:00:02.1 on minor 1
[   15.587155] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   15.587160] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[   15.587193] i915 0000:00:02.2: enabling device (0000 -> 0002)
[   15.587207] i915 0000:00:02.2: [drm] Found ALDERLAKE_S (device ID 4680) display version 12.00 stepping C0
[   15.587222] i915 0000:00:02.2: Running in SR-IOV VF mode
[   15.587648] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.20.1
[   15.588286] i915 0000:00:02.2: [drm] VT-d active for gfx access
[   15.588304] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[   15.588681] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.20.1
[   15.589019] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.20.1
[   15.589535] i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   15.589539] i915 0000:00:02.2: HuC firmware PRELOADED
[   15.591387] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[   15.591391] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[   15.591496] [drm] Initialized i915 1.6.0 for 0000:00:02.2 on minor 2
[   15.591719] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   15.591724] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   15.591728] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[   15.591759] i915 0000:00:02.3: enabling device (0000 -> 0002)
[   15.591775] i915 0000:00:02.3: [drm] Found ALDERLAKE_S (device ID 4680) display version 12.00 stepping C0
[   15.591788] i915 0000:00:02.3: Running in SR-IOV VF mode
[   15.591964] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.20.1
[   15.592210] i915 0000:00:02.3: [drm] VT-d active for gfx access
[   15.592228] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[   15.592318] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.20.1
[   15.592575] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.20.1
[... cut ...]
[   65.643469] i915 0000:00:02.6: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   65.643475] i915 0000:00:02.7: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   65.709444] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   65.709456] i915 0000:00:02.7: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   65.770445] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=io+mem:owns=io+mem
[   67.112039] i915 0000:00:02.0: Disabled 7 VFs
```

altrnate32, 2025-09-01:
Addition: It works if I do this
Remove PCI(E) device from VM
Reboot
Start VM (without PCI device)
Re-attach PCI device
Reboot VM
dmesg then shows:
```
[  246.337088] vfio-pci 0000:00:02.3: resetting
[  246.337586] i915 0000:00:02.0: VF3 FLR
[  246.442774] vfio-pci 0000:00:02.3: reset done
[  246.499090] vfio-pci 0000:00:02.3: resetting
[  246.499694] i915 0000:00:02.0: VF3 FLR
[  246.601862] vfio-pci 0000:00:02.3: reset done
```
VM start normally and device is functioning.
Did not try reboot yet (as I am happy things are running again ;) )

darkcrono, 2025-09-02:
Thank you! that seems to work, so the workaround is every time I boot the pve server would be to spin up the VM without the pcie passthrough, do it while it is running then do a reboot.

I'm now running into another issue, the linux (24.03 lts ubuntu) vm doesn't seem to see any rendering device when I do
```
$ ls -l /dev/dri
total 0
drwxr-xr-x 2 root root      60 Sep  2 02:28 by-path
crw-rw---- 1 root video 226, 0 Sep  2 02:28 card0
```

Within the VM, when I do `$ lspci -nnk` I get,
```
00:00.0 Host bridge [0600]: Intel Corporation 82G33/G31/P35/P31 Express DRAM Controller [8086:29c0]
	Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
00:01.0 VGA compatible controller [0300]: Device [1234:1111] (rev 02)
	Subsystem: Red Hat, Inc. Device [1af4:1100]
	Kernel driver in use: bochs-drm
	Kernel modules: bochs
00:1a.0 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB 
```

not sure if that is correct, but it isn't xe which is blacklisted
Also this now on a new VM which is `uname -r` is 6.8.0-79-generic,
with the VM dkms status,
`i915-sriov-dkms/2025.07.22, 6.8.0-79-generic, x86_64: installed`

Also `$vainfo` produces,
```
$ vainfo
error: can't connect to X server!
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/bochs-drm_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```

darkcrono, 2025-09-02:
thanks for the suggestion! tried the autoinstall and reinstalling as well on the VM but no luck
`dkms status` shows,
`i915-sriov-dkms/2025.07.22, 6.8.0-79-generic, x86_64: installed`

so it is installed and active on the VM. When I do `lspci -nnk` it indicates its using bochs-rm kernal drivers for the VGA controller, is that the correct one it should be using?
```
$ lspci -nnk
00:01.0 VGA compatible controller [0300]: Device [1234:1111] (rev 02)
	Subsystem: Red Hat, Inc. Device [1af4:1100]
	Kernel driver in use: bochs-drm
	Kernel modules: bochs
```

altrnate32, 2025-09-02:
> darkcrono left a comment [(strongtz/i915-sriov-dkms#342)](https://github.com/strongtz/i915-sriov-dkms/issues/342#issuecomment-3246239473)
>
> hmmm, upon further testing, it looks like the workaround of attaching the PCI VF and rebooting doesn't work.
> How are you rebooting? I run sudo reboot within the VM but it doesn't look like the PCI device gets attached when rebooting within the VM. I also hit RESET for the VM and the PCI device in hardware remains orange so it doesn't seem to get attached during a reboot
>
> So when I stop the VM and then start it with the PCI device attached it runs into the same issue before where the VFs get disabled.
>
> Proxmox error shows,
> TASK ERROR: no pci device info for device '0000:00:02.1'
>
> while dmesg | grep i915 shows,
>
> # dmesg | grep i915
> [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.14.11-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
> [    0.031722] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.14.11-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
> [    5.790481] i915: module verification failed: signature and/or required key missing - tainting kernel
> [    6.088124] i915: You are using the i915-sriov-dkms module, a ported version of the i915 module with SR-IOV support.
> [    6.088127] i915: Please file any bug report at https://github.com/strongtz/i915-sriov-dkms/issues/new.
> [    6.088128] i915: Module Homepage: https://github.com/strongtz/i915-sriov-dkms
> [    6.088456] i915 0000:00:02.0: [drm] Found ALDERLAKE_P/ADL-N (device ID 46d3) display version 13.00 stepping D0
> [... cut ...]
> [  293.101990] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
> [  293.101995] i915 0000:00:02.3: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
> [  293.177077] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=io+mem:owns=io+mem
> [  294.394071] i915 0000:00:02.0: Disabled 3 VFs

I reboot the host, not the VM.
De-attach
Shutdown VM
Reboot host
Attach
Start VM.

-------- Original Message --------

darkcrono, 2025-09-02:
Yesss!! thank you for the clarification, it finally works!

Sure is quite the odd workaround, but once its done I can now reboot my host and the VMs start up correctly with the VF attached.

Also a side note of what else helped was to not pass through the VF with the "All Functions" flag, thats probably why it always took over the HDMI port before and crashed/disabled the VFs

closed 2025-09-02 by darkcrono