strongtz/i915-sriov-dkms #343: Docs Update Request/Odd Issue on Upgrade to PVE 9: /etc/sysfs.conf.d/* Files Not Recognized Where /etc/sysfs.conf does not exist

johntdavis84, 2025-08-31:
**tl;dr: Based on the below, _and if someone else can replicate my results_, I would suggest updating the instructions for the sysfs.conf configuration to indicate that the following configuration is required for PVE 9 if you're trying to use override files in the `/etc/sysfs.d` sub directory.**

I lost my VFs after an otherwise successful upgrade from PVE to PVE 9. They were gone from `lspci`.
The DKMS module showed as installed on `dkms status`, and my kernel command line was unchanged from my PVE 8 install.

On investigation, I saw a note I'd never seen in `dmesg`: 7 VFs "could" be enabled.
As in, everything was there to turn them on, but they weren't turned on.

After thinking about it a bit, I realized that sounded like they weren't being started by sysfs, which was odd since I had a config file for that.
```
root@andromeda2:/etc/sysfs.d# ls /etc/sysfs.d/
01-intel-iGPU-sriov.conf

root@andromeda2:/etc/sysfs.d# cat /etc/sysfs.d/01-intel-iGPU-sriov.conf 
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
```

What I did _not_ have was an `/etc/sysfs.conf` file.
From my googling, that's loaded "if it exists," and then the overrides in the sub-directory are loaded.

But apparently the behavior is changed a bit in Trixie/Debian 13. It wasn't loading anything from the override directory until I created a new `/etc/sysfs.conf` file. It's blank aside from a couple of comment lines and the blank line at the end that `nano` always puts on things.

```
root@andromeda2:/etc/sysfs.d# cat /etc/sysfs.conf
# Main SysFS.conf File
# Custom entries in sysfs.conf.d subfolder.
```

One reboot later, and I was back in business.

Sauryc, 2025-08-31:
You're the one who causes the most trouble everyday, using a stupid avatar, stupid thing

Sauryc, 2025-08-31:
Is the person named SInisterPisces on the PVE forum also an idiot like you?

johntdavis84, 2025-09-05:
This seems to possibly be an expected change in Debian that's just not well-documented upstream, but I'm not completely sure about that.

I've opened a bug report with Proxmox to try to get some clarity on what's going on. After that, if necessary I'll draft a proposed change to the documentation here. Something that worked before obviously broke, but and I know one way to fix it, but I'm not sure that's the only way.

See: https://bugzilla.proxmox.com/show_bug.cgi?id=6770