strongtz/i915-sriov-dkms #351: PVE 9 and kernel 6.14.11-2 no VFs created

the0rookie, 2025-09-25:
Should this work on PVE9 with kernel 6.14.11-2? I tried it and it just doesn't create the VFs. Should I go back to kernel 6.8 in order to make it work?

johntdavis84, 2025-09-25:
Was it working before?

I was already on 6.14.10 when I upgraded to 6.14.11-2, and my setup survived.

```
root@andromeda2:~# uname -a
Linux andromeda2 6.14.11-2-pve #1 SMP PREEMPT_DYNAMIC PMX 6.14.11-2 (2025-09-12T09:46Z) x86_64 GNU/Linux

root@andromeda2:~# lspci -nnk | grep -i "vga"
00:02.0 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:02.1 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:02.2 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:02.3 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:02.4 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:02.5 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:02.6 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:02.7 VGA compatible controller [0300]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
```

What's the output of `dmesg -T | grep -i "i915"`?

the0rookie, 2025-09-25:
Here it is:

root@pve1:~# dmesg -T | grep -i "i915"
[Fri Sep 26 00:52:40 2025] Command line: BOOT_IMAGE=/boot/vmlinuz-6.14.11-2-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[Fri Sep 26 00:52:40 2025] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.14.11-2-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
root@pve1:~# 

No, I haven't tested it on PVE 8 and it is a new (well second hand LoL) machine. I got it few weeks ago with the idea to use it for OPNsense, but since it has the GPU, though I could use it for some encoding as well.

johntdavis84, 2025-09-25:
I apologize; I didn't read your message closely enough before I replied the last time.

This driver won't work with a 9th generation Intel iGPU. Those don't support SR-IOV. See: https://www.intel.com/content/www/us/en/support/articles/000093216/graphics/processor-graphics.html

Look up instructions for configuring Proxmox to use iGPUs with **Intel® Graphics Virtualization Technology –g (GVT-g)**.

the0rookie, 2025-09-25:
Ohh... OK. You saved me some hours of banging my head trying to figure out what's wrong. Thanks for the prompt reply!

closed 2025-09-25 by the0rookie