strongtz/i915-sriov-dkms #352: Unable to get VFs working on Proxmox 9.0.5 6.14.11-2-pve

ctecte, 2025-09-26:
I have first enabled secure boot before attempting to follow the steps on the ReadMe. I am using an N100 Mini PC

Kernel: 6.14.11-2-pve

I have a similar issue to #299, where the lspci only shows me one igpu. I tried following what a user replied to the issue
```echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs ```
But i was met with no file exists. "cat sriov_numvfs" prints 0. 

```
root@n100black:/sys/devices/pci0000:00/0000:00:02.0# ls -l | grep vfs
-rw-r--r-- 1 root root      4096 Sep 26 16:49 sriov_numvfs
-r--r--r-- 1 root root      4096 Sep 26 16:48 sriov_totalvfs
root@n100black:/sys/devices/pci0000:00/0000:00:02.0# echo 7 > sriov_numvfs 
-bash: echo: write error: No such file or directory
root@n100black:/sys/devices/pci0000:00/0000:00:02.0# cat sriov_totalvfs 
7
root@n100black:/sys/devices/pci0000:00/0000:00:02.0# uname -r
6.14.11-2-pve
```

sudo nano the file would not allow me to write either.

I have seen other people on issues saying that proxmox 9.0 works with dkms, but it doesnt seem like i can get mine working. 

For reference, these are the logs from the dpkg -i, and i think a related issue is the last line, 
root@n100black:~# dkms status
i915-sriov-dkms/2025.07.22, 6.14.11-2-pve, x86_64: installed (Original modules exist)

```
2025-09-26 16:12:37 (71.7 MB/s) - ‘/tmp/i915-sriov-dkms_2025.07.22_amd64.deb’ saved [1928842/1928842]

root@n100black:~# dpkg -i /tmp/i915-sriov-dkms_2025.07.22_amd64.deb
(Reading database ... 96195 files and directories currently installed.)
Preparing to unpack .../i915-sriov-dkms_2025.07.22_amd64.deb ...
Module i915-sriov-dkms/2025.07.22 for kernel 6.14.11-2-pve (x86_64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.14.11-2-pve/updates/dkms/i915.ko
Restoring archived original module /lib/modules/6.14.11-2-pve/kernel/drivers/gpu/drm/i915/i915.ko
Deleting /lib/modules/6.14.11-2-pve/updates/dkms/kvmgt.ko
Restoring archived original module /lib/modules/6.14.11-2-pve/kernel/drivers/gpu/drm/i915/kvmgt.ko
Running depmod... done.

Deleting module i915-sriov-dkms/2025.07.22 completely from the DKMS tree.
Unpacking i915-sriov-dkms (2025.07.22) over (2025.07.22) ...
Setting up i915-sriov-dkms (2025.07.22) ...
install dkms modules for all kernels
Loading new i915-sriov-dkms/2025.07.22 DKMS files...
Building for 6.14.11-2-pve

Building initial module i915-sriov-dkms/2025.07.22 for 6.14.11-2-pve
Sign command: /lib/modules/6.14.11-2-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module(s)........................................... done.
Signing module /var/lib/dkms/i915-sriov-dkms/2025.07.22/build/i915.ko
Signing module /var/lib/dkms/i915-sriov-dkms/2025.07.22/build/kvmgt.ko
Found pre-existing /lib/modules/6.14.11-2-pve/kernel/drivers/gpu/drm/i915/i915.ko, archiving for uninstallation
Installing /lib/modules/6.14.11-2-pve/updates/dkms/i915.ko
Found pre-existing /lib/modules/6.14.11-2-pve/kernel/drivers/gpu/drm/i915/kvmgt.ko, archiving for uninstallation
Installing /lib/modules/6.14.11-2-pve/updates/dkms/kvmgt.ko
Running depmod... done.
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.148.3) ...
update-initramfs: Generating /boot/initrd.img-6.14.11-2-pve
Running hook script 'zz-proxmox-boot'..
Re-executing '/etc/kernel/postinst.d/zz-proxmox-boot' in new private mount namespace..
No /etc/kernel/proxmox-boot-uuids found, skipping ESP sync.

root@n100black:~# dkms status
i915-sriov-dkms/2025.07.22, 6.14.11-2-pve, x86_64: installed (Original modules exist)
root@n100black:~#

```

admarty, 2025-09-26:
try with a rom file: https://github.com/LongQT-sea/intel-igpu-passthru

Kh3nsu, 2025-10-03:
> try with a rom file: https://github.com/LongQT-sea/intel-igpu-passthru

Does this replace the installation of the sriov driver inside the VM? Does it use the normal Intel driver then?

johntdavis84, 2025-10-05:
> > try with a rom file: https://github.com/LongQT-sea/intel-igpu-passthru
> 
> Does this replace the installation of the sriov driver inside the VM? Does it use the normal Intel driver then?

No. This is specifically for Windows 11 VMs to avoid Error 43. It won't help if Proxmox itself isn't seeing the SR-IOV PCIe devices correctly. So, let's try to get that fixed before running any VMs.

I'm assuming you're running as root.

On the Proxmox host:
1. Confirm `sysfsutils` is installed: `apt update && apt install sysfsutils`.
2. If `sysfsutils` is not installed, the above command will install it.
3. Let's make sure the sysfs commands are configured correctly. What's the output of: `ls -lah /etc | grep -i "sysfs"`?

ctecte, 2025-10-06:
> No. This is specifically for Windows 11 VMs to avoid Error 43. It won't help if Proxmox itself isn't seeing the SR-IOV PCIe devices correctly. So, let's try to get that fixed before running any [...]

[image]

On a side note, i was doing some digging and I'm not sure if my mini pc motherboard supports IOMMU. I couldn't find a definitive answer whether it is enabled or not a feature on my T9plus n100 mini pc.

johntdavis84, 2025-10-06:
Thanks for posting the output of those commands. SysFS is there, at least.

**This entire guide is worth reading, but I've linked dirctly to the commands for verifying IOMMU support: https://pve.proxmox.com/wiki/PCI_Passthrough#Verifying_IOMMU_parameters**

Also, what is the output of:
```
cat /etc/sysfs.conf
cat /etc/sysfs.d/<any file in this folder>
```

johntdavis84, 2025-11-03:
Try this:

1. Leave the `/etc/sysfs.conf` file empty, so it exists.
2. create a new file in `/etc/sysfs.d`: `10-sriov-sysfs.conf` with the content, on a single line: `devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7`
3. Reboot.
4. Check for VFs.

If this works, there's some sort of actual bug in Debian 13 or an issue with the documentation not matching the expected behavior.

TigerBeanst, 2025-11-03:
> Try this:
>
> 1. Leave the `/etc/sysfs.conf` file empty, so it exists.
> 2. create a new file in `/etc/sysfs.d`: `10-sriov-sysfs.conf` with the content, on a single line: [...]

it failed.
```
root@TigerPVE:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
root@TigerPVE:~# cd /sys/devices/pci0000:00/0000:00:02.0
root@TigerPVE:/sys/devices/pci0000:00/0000:00:02.0# ls -l | grep vfs
-rw-r--r-- 1 root root      4096 Nov  4 01:51 sriov_numvfs
-r--r--r-- 1 root root      4096 Nov  4 02:05 sriov_totalvfs
root@TigerPVE:/sys/devices/pci0000:00/0000:00:02.0# cat sriov_totalvfs
7
root@TigerPVE:/sys/devices/pci0000:00/0000:00:02.0# cat sriov_numvfs
0
```

```
root@TigerPVE:/etc/sysfs.d# lspci -s 00:02.0 -v
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics] (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: Intel Corporation Device 7270
        Flags: bus master, fast devsel, latency 0, IRQ 158, IOMMU group 0
        Memory at 60e7000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        I/O ports at 3000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, IntMsgNum 0
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Capabilities: [320] Single Root I/O Virtualization (SR-IOV)
        Kernel driver in use: vfio-pci
        Kernel modules: i915, xe
```

Kh3nsu, 2025-11-03:
@TigerBeanst please share your output from:

`cat /proc/cmdline`

0xCA71, 2025-12-15:
Hi all,
I found that I have the same problem with @TigerBeanst 's 'Edit 2' section, my case is N100 + PVE 9.1.1.

**Maybe there something wrong when starting VM.**

My `cat /proc/cmdline` and `lspci -nnk -s 00:02.0` is: ( nothing special )
```
root@virtual:~# cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-6.17.2-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
root@virtual:~# lspci -nnk -s 00:02.0
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [UHD Graphics] [8086:46d1]
	DeviceName: Onboard - Video
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: vfio-pci
	Kernel modules: i915, xe
```
`lspci | grep VGA` and `dmesg | grep i915`  when just start up: ( looks like normal )
```
root@virtual:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]  
00:02.1 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics] 
00:02.2 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics] 
00:02.3 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
root@virtual:~# dmesg | grep i915
... <truncated>
[   13.987946] i915 0000:00:02.4: Running in SR-IOV VF mode
[   13.990448] i915 0000:00:02.4: [drm] GT0: GUC: interface version 0.1.17.0
[   13.991122] i915 0000:00:02.4: [drm] VT-d active for gfx access
[   13.991156] i915 0000:00:02.4: [drm] Using Transparent Hugepages
[   13.991349] i915 0000:00:02.4: [drm] GT0: GUC: interface version 0.1.17.0
[   13.991874] i915 0000:00:02.4: [drm] GT0: GUC: interface version 0.1.17.0
[   13.992548] i915 0000:00:02.4: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   13.992552] i915 0000:00:02.4: HuC firmware PRELOADED
[   13.995273] i915 0000:00:02.4: [drm] Protected Xe Path (PXP) protected content support initialized
[   13.995279] i915 0000:00:02.4: [drm] PMU not supported for this GPU.
[   13.996443] [drm] Initialized i915 1.6.0 for 0000:00:02.4 on minor 4
[   13.996591] i915 0000:00:02.0: Enabled 4 VFs

```
But after I add one VF as a PCI device to any VM ( I tried Windows10 and Ubuntu22.04, both fail ), then start the VM, PVE will report a error like this: **`TASK ERROR: no pci device info for device '0000:00:02.1'`**

**Then all VFs disappears**, `lspci | grep VGA` displays this:
```
root@virtual:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
```
and these messages was append to `dmesg` :
```
root@virtual:~# dmesg | grep i915
... <truncated>
[11341.036543] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[11341.036554] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[11341.036558] i915 0000:00:02.3: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[11341.036562] i915 0000:00:02.4: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[11341.109562] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[11341.109570] i915 0000:00:02.3: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[11341.109573] i915 0000:00:02.4: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[11341.155540] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[11341.155548] i915 0000:00:02.4: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[11341.200563] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=io+mem:owns=io+mem
[11342.434238] i915 0000:00:02.0: Disabled 4 VFs
root@virtual:~# dmesg | grep vfio-pci
[11342.566097] vfio-pci 0000:00:02.0: vgaarb: deactivate vga console
[11342.566104] vfio-pci 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[11342.589397] vfio-pci 0000:00:02.0: resetting
[11342.693520] vfio-pci 0000:00:02.0: reset done
```
**Probably we need find why the VFs disabled?**
Feel free to ask me for more, I am glad to test for new solutions.

whao, 2026-02-17:
I am encountering the same issue.
```
root@proxmox:~# cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-6.17.9-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
root@proxmox:~# dkms status
i915-sriov-dkms/2026.02.09, 6.17.9-1-pve, x86_64: installed (Original modules exist)
root@proxmox:~# cat /etc/sysfs.d/10-sriov-sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
```
But no VFs showed in `lspci`
```
root@proxmox:~# lspci
00:00.0 Host bridge: Intel Corporation Device 4e24
00:02.0 VGA compatible controller: Intel Corporation JasperLake [UHD Graphics] (rev 01)
00:04.0 Signal processing controller: Intel Corporation Dynamic Tuning service
00:14.0 USB controller: Intel Corporation Jasper Lake USB 3.1 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Jasper Lake Shared SRAM (rev 01)
00:16.0 Communication controller: Intel Corporation Management Engine Interface (rev 01)
00:17.0 SATA controller: Intel Corporation Jasper Lake SATA AHCI Controller (rev 01)
00:1c.0 PCI bridge: Intel Corporation Jasper Lake PCIe Root Port #1 (rev 01)
00:1c.4 PCI bridge: Intel Corporation Jasper Lake PCIe Root Port #5 (rev 01)
00:1c.5 PCI bridge: Intel Corporation Device 4dbd (rev 01)
00:1c.6 PCI bridge: Intel Corporation Jasper Lake PCIe Root Port #7 (rev 01)
00:1c.7 PCI bridge: Intel Corporation Device 4dbf (rev 01)
00:1f.0 ISA bridge: Intel Corporation Jasper Lake eSPI Controller (rev 01)
00:1f.3 Audio device: Intel Corporation Jasper Lake HD Audio (rev 01)
00:1f.4 SMBus: Intel Corporation Jasper Lake SMBus (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Jasper Lake SPI Controller (rev 01)
01:00.0 Non-Volatile memory controller: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1202 (DRAM-less) (rev 01)
02:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
03:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
04:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
05:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
```
No `sriov_numvfs` is created in `sys`:
```
root@proxmox:~# echo 7 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs: Permission denied
```

pojlFDlxCOvZ4Kg8y1l4, 2026-09-05:
> I got this to work on N355, and N150 PCs. However, the instructions on the main page did not cover my edge case.
> 
> NOTE: Get the device ID, omit 0x. `cat /sys/devices/pci0000:00/0000:00:02.0/device` 46d3 - n355 46d1 - n150
> 
> 1. Use the new XE driver.
> 
> /etc/default/grub `GRUB_CMDLINE_LINUX_DEFAULT="intel_iommu=on xe.max_vfs=7 xe.force_probe=46d3 module_blacklist=i915"`
> 
> 2. grub edit only seems ineffective, even after update-initramfs -u -k all, and update-grub, the VGA driver in lspci is still on i915.
>    a) append /etc/kernel/cmdline
>    `intel_iommu=on xe.max_vfs=7 xe.force_probe=46d3 module_blacklist=i915`
>    example: `root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on xe.max_vfs=7 xe.force_probe=46d3 module_blacklist=i915`
> 3. initramfs -u -k all, update-grub, reboot, then manually add the functions
>    `echo 7 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`
> 
> After this, I was able to see the 7 VFs in proxmox, and assign them a resource group. The key factors missing from the guide seemed to be XE for this Alder-Lake platform, and the cmdline edit. If you've done other hacks to try to get this to work, you might have to revert your changes, or start fresh.
> 
> If you'd like to see a video demo of it working, please leave a thumbs up

switching to XE also worked for me. i915 would just not enable SR-IOV, even when adding the certificates for secure boot. I assume those certs were still necessary though, secure boot is enabled. i5-12400 and Proxmox 9.2.11 with kernel 7.0.14-15-pve. 
I recall this being a thing also on my N100 system, but this is from half a year ago. I915 not doing the thing but Xe did. Sadly I did not take notes of it.